micro-ROS/micro_ros_platformio #152: Build using github action suddenly fails

yairshlomi1, 2024-09-29:
## Issue template

- Hardware description: Teensy 4.0
- RTOS: Teensy 4.0, ros2 humble
- Installation type: github action
- Version or commit hash: 25ab6593ad0c25ce32342a86fa1ac3d7998f90f0

#### Steps to reproduce the issue
Hi. I'm using github action to build my micro-ros platformio project.
I've been working with this specific github action a few months without any changes, and today it suddenly stopped working.
I've tested it on old tags in my repository that was successfully built using the action in the past and the new builds fails.
**on my ubuntu 22.04 machine it still builds without any errors or warnings** 

this is the action file:
```yaml
name: Build

on: [push]

jobs:
  build:
    runs-on: ubuntu-latest    

    steps:
    - uses: actions/checkout@v4
    - name: Cache pip
      uses: actions/cache@v4
      with:
        path: ~/.cache/pip
        key: ${{ runner.os }}-pip-${{ hashFiles('**/requirements.txt') }}
        restore-keys: |
          ${{ runner.os }}-pip-
    - name: Cache PlatformIO
      uses: actions/cache@v4
      with:
        path: ~/.platformio
        key: ${{ runner.os }}-${{ hashFiles('**/lockfiles') }}

    - name: Set up Python
      uses: actions/setup-python@v5
      with:
        python-version: '3.10'

    - name: Install PlatformIO
      run: |
        sudo apt-get install -y python3.10-venv
        wget -O get-platformio.py https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py
        python3 get-platformio.py

        
    - name: Run PlatformIO
      run: |
        export PATH="$HOME/.platformio/penv/bin:$PATH"
        pio run 
    
    - name: Archive artifacts 
      id : archive_artifacts    
      uses: actions/upload-artifact@v4   
      with:
        name: ninox-gimbal-mcu-firmware
        path: .pio/build/**/*.hex  
           
    - name: Create Release
      if: startsWith(github.ref, 'refs/tags/')
      id: create_release
      uses: softprops/action-gh-release@v2
      env:
        GITHUB_TOKEN: ${{ secrets.GITHUB_TOKEN }}
      with:
        token: ${{ secrets.GITHUB_TOKEN }}
[... 292 lines cut ...]
    File "/home/runner/work/led_publisher/led_publisher/.pio/libdeps/teensy40/micro_ros_platformio/build/mcu/install/lib/python3.10/site-packages/rosidl_adapter/__main__.py", line 19, in <module>
      sys.exit(main())
    File "/home/runner/work/led_publisher/led_publisher/.pio/libdeps/teensy40/micro_ros_platformio/build/mcu/install/lib/python3.10/site-packages/rosidl_adapter/main.py", line 53, in main
      abs_idl_file = convert_to_idl(
    File "/home/runner/work/led_publisher/led_publisher/.pio/libdeps/teensy40/micro_ros_platformio/build/mcu/install/lib/python3.10/site-packages/rosidl_adapter/__init__.py", line 19, in convert_to_idl
      return convert_msg_to_idl(
    File "/home/runner/work/led_publisher/led_publisher/.pio/libdeps/teensy40/micro_ros_platformio/build/mcu/install/lib/python3.10/site-packages/rosidl_adapter/msg/__init__.py", line 39, in convert_msg_to_idl
      expand_template('msg.idl.em', data, output_file, encoding='iso-8859-1')
    File "/home/runner/work/led_publisher/led_publisher/.pio/libdeps/teensy40/micro_ros_platformio/build/mcu/install/lib/python3.10/site-packages/rosidl_adapter/resource/__init__.py", line 23, in expand_template
      content = evaluate_template(template_name, data)
    File "/home/runner/work/led_publisher/led_publisher/.pio/libdeps/teensy40/micro_ros_platformio/build/mcu/install/lib/python3.10/site-packages/rosidl_adapter/resource/__init__.py", line 69, in evaluate_template
      _interpreter.shutdown()

  AttributeError: 'NoneType' object has no attribute 'shutdown'

Call Stack (most recent call first):
  /home/runner/work/led_publisher/led_publisher/.pio/libdeps/teensy40/micro_ros_platformio/build/mcu/install/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:130 (rosidl_adapt_interfaces)
  CMakeLists.txt:16 (rosidl_generate_interfaces)

---
Failed   <<< builtin_interfaces [1.15s, exited with code 1]

========================= [FAILED] Took 87.49 seconds =========================
```
#### Additional information

jeevan-vexev, 2024-09-30:
:+1:  am having the same issues too :( First noticed it 27 Sep if that helps anyone!

yairshlomi1, 2024-09-30:
presumably, there is a new pip version that parses and installs colcon-common-extensions differently, which installs EmPy version 4.x.x, when micro-ros needs EmPy version higher than 3.3.4 but apparently lower than 4.x.x. Adding another step (in the jobs section in the action file) that manually installs the correct version of EmPy inside the venv (created by platformio) before calling to ```pio run``` solves this issue.

```yaml
- name: Install empy in PlatformIO Environment
    run: |
        ~/.platformio/penv/bin/pip install --upgrade empy==3.3.4

```

closed 2024-09-30 by yairshlomi1

roboticvedant, 2024-10-29:
i am still having the same issue