micro-ROS/micro_ros_platformio #153: Build failing for humble in vscode - clock_gettime missing

mattwilliamson, 2024-09-30:
## Issue template

- Hardware description: raspberry pi pico
- RTOS: freertos
- Installation type: platformio
- Version or commit hash: humble

#### Steps to reproduce the issue
Hit build in vscode on platformio

platformio.ini
```ini
[platformio]
default_envs = pico

[env]
framework = arduino
lib_deps = 
	https://github.com/micro-ROS/micro_ros_platformio
        https://github.com/end2endzone/AnyRtttl
	sparkfun/SparkFun 9DoF IMU Breakout - ICM 20948 - Arduino Library@^1.2.12
	adafruit/Adafruit NeoPixel@^1.12.3

lib_archive = false
board_microros_distro = humble
board_microros_transport = serial

; Enable Sensor Fusion on the IMU
build_flags = 
	-DICM_20948_USE_DMP

[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
board_build.core = earlephilhower
board_build.mcu = rp2040
board_build.f_cpu = 133000000L

upload_protocol = cmsis-dap
debug_tool = cmsis-dap
debug_speed = 5000
upload_speed = 5000
monitor_speed = 115200
build_type = debug

; Examples 1: disable initial breakpoint
debug_init_break =
```

#### Expected behavior

Builds successfully. This same project worked a couple months ago before re-cloning it.

#### Actual behavior

```
Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; board: pico; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.14.0+sha.2a8c40e) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (blackmagic, cmsis-dap, jlink, pico-debug, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.40003.0+sha.6acf03e 
 - tool-picotool-rp2040-earlephilhower @ 5.140200.240926 (14.2.0) 
 - toolchain-rp2040-earlephilhower @ 5.140200.240926 (14.2.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing importlib-resources with pip at PlatformIO environment
/home/matt/.platformio/penv/bin/python -m pip install importlib-resources
Requirement already satisfied: importlib-resources in /home/matt/.platformio/penv/lib/python3.10/site-packages (6.4.5)
Installing pyyaml with pip at PlatformIO environment
/home/matt/.platformio/penv/bin/python -m pip install pyyaml
Requirement already satisfied: pyyaml in /home/matt/.platformio/penv/lib/python3.10/site-packages (6.0.2)
Installing markupsafe==2.0.1 with pip at PlatformIO environment
/home/matt/.platformio/penv/bin/python -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in /home/matt/.platformio/penv/lib/python3.10/site-packages (2.0.1)
Configuring pico with transport serial
Downloading micro-ROS dev dependencies
         - Downloaded ament_cmake
         - Downloaded ament_lint
         - Downloaded ament_package
         - Downloaded googletest
         - Downloaded ament_cmake_ros
         - Downloaded ament_index
Building micro-ROS dev dependencies
Downloading micro-ROS library
         - Downloaded microcdr
         - Downloaded microxrcedds_client
         - Downloaded rcl_lifecycle
         - Downloaded rcl
         - Downloaded rcl_yaml_param_parser (ignored)
         - Downloaded rcl_action
         - Downloaded rclc_examples (ignored)
         - Downloaded rclc
         - Downloaded rclc_lifecycle
         - Downloaded rclc_parameter
         - Downloaded micro_ros_utilities
         - Downloaded rcutils
         - Downloaded micro_ros_msgs
         - Downloaded rmw_microxrcedds
         - Downloaded rosidl_typesupport_cpp (ignored)
         - Downloaded rosidl_typesupport_c
         - Downloaded rosidl_typesupport_microxrcedds_cpp (ignored)
         - Downloaded rosidl_typesupport_microxrcedds_cpp_tests
         - Downloaded rosidl_typesupport_microxrcedds_c_tests
         - Downloaded rosidl_typesupport_microxrcedds_test_msg
         - Downloaded rosidl_typesupport_microxrcedds_c
         - Downloaded rosidl_runtime_cpp (ignored)
         - Downloaded rosidl_runtime_c
         - Downloaded rosidl_generator_c
         - Downloaded rosidl_typesupport_introspection_tests
         - Downloaded rosidl_typesupport_introspection_cpp (ignored)
         - Downloaded rosidl_typesupport_interface
         - Downloaded rosidl_parser
         - Downloaded rosidl_cmake
         - Downloaded rosidl_cli
         - Downloaded rosidl_adapter
         - Downloaded rosidl_typesupport_introspection_c
         - Downloaded rosidl_generator_cpp (ignored)
         - Downloaded rmw_implementation_cmake
         - Downloaded rmw
         - Downloaded rosgraph_msgs
         - Downloaded test_msgs
         - Downloaded lifecycle_msgs
         - Downloaded composition_interfaces
         - Downloaded builtin_interfaces
         - Downloaded rcl_interfaces
         - Downloaded action_msgs
         - Downloaded statistics_msgs
         - Downloaded rosidl_default_generators
         - Downloaded rosidl_default_runtime
         - Downloaded unique_identifier_msgs
         - Downloaded geometry_msgs
         - Downloaded stereo_msgs
         - Downloaded sensor_msgs_py
         - Downloaded visualization_msgs
         - Downloaded common_interfaces
         - Downloaded shape_msgs
         - Downloaded std_srvs
         - Downloaded nav_msgs
         - Downloaded sensor_msgs
         - Downloaded actionlib_msgs
         - Downloaded diagnostic_msgs
         - Downloaded std_msgs
         - Downloaded trajectory_msgs
         - Downloaded test_interface_files
         - Downloaded test_rmw_implementation
         - Downloaded rmw_implementation
         - Downloaded rcl_logging_noop
         - Downloaded rcl_logging_spdlog (ignored)
         - Downloaded rcl_logging_interface
         - Downloaded test_tracetools
         - Downloaded ros2trace
         - Downloaded tracetools_launch
         - Downloaded test_tracetools_launch
         - Downloaded tracetools_trace
         - Downloaded tracetools_read
         - Downloaded tracetools
         - Downloaded tracetools_test
         - Extra packages folder not found, skipping...
Building micro-ROS library
Build mcu micro-ROS environment failed: 
--- stderr: microxrcedds_client
/home/matt/src/deepdrive_micro/.pio/libdeps/pico/micro_ros_platformio/build/mcu/src/Micro-XRCE-DDS-Client/src/c/util/time.c: In function 'uxr_nanos':
/home/matt/src/deepdrive_micro/.pio/libdeps/pico/micro_ros_platformio/build/mcu/src/Micro-XRCE-DDS-Client/src/c/util/time.c:71:5: error: implicit declaration of function 'clock_gettime' [-Wimplicit-function-declaration]
   71 |     clock_gettime(CLOCK_REALTIME, &ts);
      |     ^~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/microxrcedds_client.dir/build.make:272: CMakeFiles/microxrcedds_client.dir/src/c/util/time.c.obj] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/microxrcedds_client.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
---
Failed   <<< microxrcedds_client [0.27s, exited with code 2]
Error calling StartServiceByName for org.freedesktop.Notifications: Process org.freedesktop.Notifications exited with status 1

========================================================= [FAILED] Took 35.89 seconds =========================================================

```

#### Additional information

hippo5329, 2024-10-02:
Please check pull request #156.

[env:pico]
platform = raspberrypi
board = pico
lib_ldf_mode = chain+

Please check wiki https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki.

Clean all and build again.

roico-oskar, 2024-10-08:
@hippo5329 Unfortunately the issue is not resolved 

- ROS2 version: Humble
- Hardware: adafruit feather (CAN version)
- Issue: 
`error: implicit declaration of function 'clock_gettime' [-Wimplicit-function-declaration]
   74 |   clock_gettime(CLOCK_REALTIME, &timespec_now);
      |   ^~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/rcutils.dir/build.make:418: CMakeFiles/rcutils.dir/src/time_unix.c.obj] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:192: CMakeFiles/rcutils.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< rcutils [1.87s, exited with code 2]
`
Environment:

`[env:adafruit_feather_can]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git 
board = adafruit_feather_can
board_build.core = earlephilhower
board_microros_distro = humble
framework = arduino
board_microros_user_meta = roico.meta
lib_deps = 
	https://github.com/micro-ROS/micro_ros_platformio`

martincerven, 2024-10-09:
@mattwilliamson @hippo5329  was micro_ros, platformio and earle core working together at some point? I started only recently and also got errors using PIO+ Portenta H7 🥲 Only thing that is sucessfully uploading is arduino micro ros + Portenta, but that's different repo...

hippo5329, 2024-10-10:
1. Portenta H7 m7 core is supported.

[env:portenta_h7_m7]
platform = ststm32
board = portenta_h7_m7

2. I will try to add clock_gettime to the arduino pico. You may add it yourself as a temporary fix.

https://github.com/micro-ROS/micro_ros_platformio/blob/main/platform_code/arduino/clock_gettime.cpp

ARM stopped mbed support. Things might change soon.

hippo5329, 2024-10-10:
The clock_gettime issue is caused by recent toolchain update. You may revert to older version.

hippo5329, 2024-10-10:
You should use the stable version as I suggested earlier. board=pico should work for Adafruit feather can.

martincerven, 2024-10-21:
@mattwilliamson You can get older versions of Earle's core by specifying commit:

```
platform = https://github.com/maxgerhardt/platform-raspberrypi.git#196d31bbafaf60b84751b1a415d8dca2365debdf
```
This one is before adding rp2350 and Pico SDK 2.0, maybe there were some breaking changes.

mysteriousCzrsX, 2024-11-28:
@mattwilliamson Apparently #156 was supposed to fix this problem but I am experiencing the same behaviour despite the patch being applied. Did you find any other solutions apart from downgrading platform-rassberypi.

hippo5329, 2024-12-09:
I resolved this issue and created a pull-request.
https://github.com/earlephilhower/pico-quick-toolchain/pull/66

hippo5329, 2024-12-10:
```
# install toolchains
cd ~/micro_ros_arduino_examples_platformio/micro-ros_publisher
pio pkg install -e pico2
# patch time.h
cd ~
curl -O https://github.com/hippo5329/pico-quick-toolchain/raw/refs/heads/master/patches/lib-posix-timers.patch
cd ~/.platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include
patch -p4 < ~/lib-posix-timers.patch
# build and upload
cd ~/micro_ros_arduino_examples_platformio/micro-ros_publisher
pio run -e pico2 -t upload
```

hippo5329, 2024-12-10:
Same for pico.