micro-ROS/micro_ros_platformio #154: Can't Build Basic Project (Might Be Related to rosidl)

akvavit01, 2024-10-01:
## Issue template

- Hardware description: ESP WROOM 32 
- RTOS: None 
- Installation type: PlatformIO VS Code 
- Version or commit hash: jazzy 

Basically, I was just trying to build the environment first. I still hasn't used any micro-ROS-specific code in my code (it's still the default code from building a new PlatformIO project). The thing is, the build is not successful until now. I have also checked [this](https://github.com/micro-ROS/micro_ros_platformio/issues/115) issue; I have unsourced and reboot my machine, to no avail.

All I can understand is that this issue is related to rosidl, seen from the log I pasted below.

#### Steps to reproduce the issue

1. Use this platformio.ini configuration
```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio
board_microros_distro = jazzy
board_microros_transport = serial
```

2. Build

#### Expected behavior

Builds perfectly

#### Actual behavior

Build unable to complete

#### Additional information

Error Log:

CMake Error at /home/leochen/ros2_tutorial/micro_ros/tutorial/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/build/mcu/install/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:57 (message):
  execute_process(/home/leochen/.platformio/penv/bin/python -m rosidl_adapter
  --package-name builtin_interfaces --arguments-file
  /home/leochen/ros2_tutorial/micro_ros/tutorial/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/build/mcu/build/builtin_interfaces/rosidl_adapter__arguments__builtin_interfaces.json
  --output-dir
  /home/leochen/ros2_tutorial/micro_ros/tutorial/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/build/mcu/build/builtin_interfaces/rosidl_adapter/builtin_interfaces
  --output-file
  /home/leochen/ros2_tutorial/micro_ros/tutorial/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/build/mcu/build/builtin_interfaces/rosidl_adapter/builtin_interfaces.idls)
  returned error code 1:

  AttributeError processing template 'msg.idl.em'

  Traceback (most recent call last):

    File "/home/leochen/ros2_tutorial/micro_ros/tutorial/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/build/mcu/install/lib/python3.12/site-packages/rosidl_adapter/resource/__init__.py", line 51, in evaluate_template
      em.BUFFERED_OPT: True,
      ^^^^^^^^^^^^^^^

  AttributeError: module 'em' has no attribute 'BUFFERED_OPT'

  During handling of the above exception, another exception occurred:

  Traceback (most recent call last):

    File "<frozen runpy>", line 198, in _run_module_as_main
    File "<frozen runpy>", line 88, in _run_code
    File "/home/leochen/ros2_tutorial/micro_ros/tutorial/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/build/mcu/install/lib/python3.12/site-packages/rosidl_adapter/__main__.py", line 19, in <module>
      sys.exit(main())
               ^^^^^^
    File "/home/leochen/ros2_tutorial/micro_ros/tutorial/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/build/mcu/install/lib/python3.12/site-packages/rosidl_adapter/main.py", line 53, in main
      abs_idl_file = convert_to_idl(
                     ^^^^^^^^^^^^^^^
    File "/home/leochen/ros2_tutorial/micro_ros/tutorial/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/build/mcu/install/lib/python3.12/site-packages/rosidl_adapter/__init__.py", line 19, in convert_to_idl
      return convert_msg_to_idl(
             ^^^^^^^^^^^^^^^^^^^
    File "/home/leochen/ros2_tutorial/micro_ros/tutorial/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/build/mcu/install/lib/python3.12/site-packages/rosidl_adapter/msg/__init__.py", line 39, in convert_msg_to_idl
      expand_template('msg.idl.em', data, output_file, encoding='iso-8859-1')
    File "/home/leochen/ros2_tutorial/micro_ros/tutorial/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/build/mcu/install/lib/python3.12/site-packages/rosidl_adapter/resource/__init__.py", line 23, in expand_template
      content = evaluate_template(template_name, data)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/home/leochen/ros2_tutorial/micro_ros/tutorial/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/build/mcu/install/lib/python3.12/site-packages/rosidl_adapter/resource/__init__.py", line 69, in evaluate_template
      _interpreter.shutdown()
      ^^^^^^^^^^^^^^^^^^^^^

  AttributeError: 'NoneType' object has no attribute 'shutdown'

Call Stack (most recent call first):
  /home/leochen/ros2_tutorial/micro_ros/tutorial/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/build/mcu/install/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:132 (rosidl_adapt_interfaces)
  CMakeLists.txt:19 (rosidl_generate_interfaces)

akvavit01, 2024-10-01:
So, I tried with PlatfromIO core with another platformio.ini file
```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps = https://github.com/micro-ROS/micro_ros_platformio
board_microros_distro = jazzy
board_microros_transport = serial

[env:rpi_pico]
platform = raspberrypi
board = pico
framework = arduino
lib_deps = https://github.com/micro-ROS/micro_ros_platformio
board_microros_distro = jazzy
board_microros_transport = serial
```

I ran `pio run -e rpi_pico`, only to meet the same result

pablogs9, 2024-10-01:
Take a look here: https://github.com/micro-ROS/micro_ros_platformio/pull/155 and let me know if it solves the issue

closed 2024-10-01 by akvavit01