micro-ROS/micro_ros_platformio #158: Stuck in rclc_support_init

akvavit01, 2024-10-18:
## Issue template

- Hardware description: ESP32 DEVKIT V1 
- RTOS: None 
- Installation type: This repo, PlatformIO
- Version or commit hash: jazzy 

#### Steps to reproduce the issue

Use this platformio.ini
'''
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps =
  waspinator/AccelStepper@^1.64
  https://github.com/micro-ROS/micro_ros_platformio
board_microros_distro = jazzy
board_microros_transport = wifi
monitor_speed = 115200
'''

Also, use this code, in main.cpp
'''
// #include <AccelStepper.h>
#include <Arduino.h>
#include <micro_ros_platformio.h>
// #include <mission_interfaces/msg/steps.h>
#include <rcl/rcl.h>
#include <rclc/executor.h>
#include <rclc/rclc.h>
#include <std_msgs/msg/int32.h>

/**
 * @def RCCHECK(fn)
 * @brief Macro-based function to check for rcl
 * executions. If the return value is not
 * RCL_RET_OK (which means something is wrong),
 * immediately go into a blocking loop that does
 * nothing forever.
 *
 */
#define RCCHECK(fn)                \
  {                                \
    rcl_ret_t temp_rc = fn;        \
    if ((temp_rc != RCL_RET_OK)) { \
      while (true) {               \
      }                            \
    }                              \
  }

/**
 * @def RCSOFTCHECK(fn)
 * @brief Macro-based function to check for rcl
 * executions. If the return value is not
 * RCL_RET_OK (which means something is wrong),
 * does nothing. One could say that this function
 * is just to contain return value from uROS
 * execution, to prevent compiler warning.
 *
 */
#define RCSOFTCHECK(fn)            \
  {                                \
    rcl_ret_t temp_rc = fn;        \
    if ((temp_rc != RCL_RET_OK)) { \
    }                              \
  }

// AccelStepper stepper(AccelStepper::DRIVER, 22,
//                      23);

rcl_allocator_t allocator;
rclc_support_t support;
rcl_node_t node;
rclc_executor_t executor;
rcl_subscription_t pos_cmd_sub;

std_msgs__msg__Int32 pos_cmd_msg;

void PosCmdCallback(const void *msg) {
  const std_msgs__msg__Int32 *pos_cmd_msg =
      (const std_msgs__msg__Int32 *)msg;

  char log_msg[60]{};
  snprintf(log_msg, sizeof(log_msg),
           "Position command: %i",
           pos_cmd_msg->data);
  // logger::Log(logger::LogVerbosityLevel::kDebug,
  //             log_msg);
  Serial.println(log_msg);
}

void setup() {
  Serial.begin(115200);
  Serial.println(
      (char *)("Serial debug port ready!"));

  // Set up LED
  uint8_t led_pin{2};
  pinMode(led_pin, OUTPUT);
  digitalWrite(led_pin, HIGH);

  delay(1000);

  digitalWrite(led_pin, LOW);

  delay(1000);

  digitalWrite(led_pin, HIGH);

  delay(1000);

  // WiFi transport
  set_microros_wifi_transports(
      (char *)("M2M-Network"),
      (char *)("TritronikAJ9"),
      IPAddress(192, 168, 128, 72), 8888);
  Serial.println(
      (char *)("WiFi transport ready!!!"));

  allocator = rcl_get_default_allocator();

  // Create init options
  RCCHECK(rclc_support_init(&support, 0, NULL,
                            &allocator));
  Serial.println((char *)("Init options done!"));

  // Create node
  RCCHECK(rclc_node_init_default(
      &node, "controller", "", &support));
  Serial.println((char *)("Node ready!!!"));

  // Create executor
  executor =
      rclc_executor_get_zero_initialized_executor();
  RCCHECK(rclc_executor_init(&executor,
                             &support.context, 6,
                             &allocator));
  Serial.println((char *)("Executor ready!!!"));

  // Create position command subscriber
  RCCHECK(rclc_subscription_init_default(
      &pos_cmd_sub, &node,
      ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg,
                                  Int32),
      "absolute_position_command"));

  RCCHECK(rclc_executor_add_subscription(
      &executor, &pos_cmd_sub, &pos_cmd_msg,
      &PosCmdCallback, ON_NEW_DATA));
  Serial.println((char *)("Subscriber ready!!!"));
}

void loop() {
  RCSOFTCHECK(rclc_executor_spin_some(
      &executor, RCL_MS_TO_NS(100)));
}

'''

#### Expected behavior

Code runs perfectly, able to subscribe to a topic.

#### Actual behavior

Stuck here
[image]

It is crystal clear that somehow my code is stuck in rclc_support_init.

#### Additional information

Perhaps, is there a commit that changes the repo's behaviour?

hippo5329, 2024-10-18:
Please follow this wiki, https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki#micro_ros_arduino_examples_platformio

akvavit01, 2024-10-18:
Wow, very comprehensive. Thanks
Will update the results later

akvavit01, 2024-10-18:
It works, thanks
Closing this issue then

closed 2024-10-18 by akvavit01