micro-ROS/micro_ros_platformio #160: Adding a parameter server doesn't work ?

l4es, 2024-11-10:
## Issue template

- Hardware description: raspberry pi pico rp2040
- RTOS: None
- Installation type: platformio
- Version or commit hash: humble

#### Steps to reproduce the issue
N/A

#### Expected behavior
- Working with 2 publishers,1 subscription and 1 timer as previously,
- Working with the additional parameter server of 2 parameters.

#### Actual behavior
Adding the parameter server causes multiple **object deleted** , repeated **session closed** / **session established** as soon as  **datawriter created** and no more topic published : 

```
[1731234790.341469] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x63680BBD, datawriter_id: 0x002(5), publisher_id: 0x002(3)
[1731234790.341500] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x63680BBD, len: 14, data: 
0000: 81 80 0D 00 05 01 06 00 00 19 00 25 00 00
[1731234790.341515] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x63680BBD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 0E 00 00 00 80
[1731234790.342558] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x63680BBD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 0E 00 00 00 80
[1731234790.347052] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x63680BBD, len: 12, data: 
0000: 81 01 00 00 03 01 04 00 00 1A 00 02
[1731234790.347077] debug    | ProxyClient.cpp    | delete_object_unlock     | object deleted         | client_key: 0x63680BBD, object_id: 0x0000
[1731234790.347103] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x63680BBD, len: 14, data: 
0000: 81 01 00 00 05 01 06 00 00 1A 00 02 00 00
[1731234790.347480] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x63680BBD, len: 20, data: 
0000: 81 01 01 00 03 01 04 00 00 1B 00 05 03 01 04 00 00 1C 00 03
[1731234790.347702] debug    | ProxyClient.cpp    | delete_object_unlock     | object deleted         | client_key: 0x63680BBD, object_id: 0x0000
[1731234790.347731] debug    | ProxyClient.cpp    | delete_object_unlock     | object deleted         | client_key: 0x63680BBD, object_id: 0x0000
[1731234790.347740] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x63680BBD, len: 14, data: 
0000: 81 01 01 00 05 01 06 00 00 1B 00 05 00 00
[1731234790.347805] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x63680BBD, len: 14, data: 
0000: 81 01 02 00 05 01 06 00 00 1C 00 03 00 00
[1731234790.347824] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x63680BBD, len: 12, data: 
0000: 81 01 02 00 03 01 04 00 00 1D 00 12
[1731234790.347846] debug    | ProxyClient.cpp    | delete_object_unlock     | object deleted         | client_key: 0x63680BBD, object_id: 0x0001
[1731234790.347874] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x63680BBD, len: 14, data: 
0000: 81 01 03 00 05 01 06 00 00 1D 00 12 00 00
[1731234790.348236] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x63680BBD, len: 20, data: 
0000: 81 01 03 00 03 01 04 00 00 1E 00 15 03 01 04 00 00 1F 00 13
[1731234790.348363] debug    | ProxyClient.cpp    | delete_object_unlock     | object deleted         | client_key: 0x63680BBD, object_id: 0x0001
[1731234790.348390] debug    | ProxyClient.cpp    | delete_object_unlock     | object deleted         | client_key: 0x63680BBD, object_id: 0x0001
[1731234790.348428] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x63680BBD, len: 14, data: 
0000: 81 01 04 00 05 01 06 00 00 1E 00 15 00 00
[1731234790.348462] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x63680BBD, len: 14, data: 
0000: 81 01 05 00 05 01 06 00 00 1F 00 13 00 00
[1731234790.348868] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x63680BBD, len: 24, data: 
0000: 81 01 04 00 08 01 10 00 00 20 00 06 00 00 00 01 00 00 00 00 00 00 00 00
[1731234790.349181] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x63680BBD, len: 12, data: 
0000: 81 01 05 00 03 01 04 00 00 21 00 22
[1731234790.349554] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x63680BBD, len: 20, data: 
0000: 81 01 06 00 03 01 04 00 00 22 00 06 03 01 04 00 00 23 00 04
[1731234790.350119] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x63680BBD, len: 24, data: 
0000: 81 01 07 00 08 01 10 00 00 24 00 08 00 00 00 01 00 00 00 00 00 00 00 00
[1731234790.350349] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x63680BBD, len: 12, data: 
0000: 81 01 08 00 03 01 04 00 00 25 00 08
[1731234790.350650] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x63680BBD, len: 12, data: 
0000: 81 01 09 00 03 01 04 00 00 26 00 32
[1731234790.350975] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x63680BBD, len: 20, data: 
0000: 81 01 0A 00 03 01 04 00 00 27 00 25 03 01 04 00 00 28 00 23
[1731234790.352838] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x63680BBD, len: 12, data: 
0000: 81 01 0B 00 03 01 04 00 00 29 00 01
[1731234790.353243] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x63680BBD, len: 12, data: 
0000: 81 00 00 00 03 01 04 00 00 02 FF FE
[1731234790.354955] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x63680BBD, len: 16, data: 
[... 189 lines cut ...]
  RCCHECK(rclc_add_parameter_constraint_integer(&param_server, framerate_name, 0, 40, 10));
  RCCHECK(rclc_add_parameter_constraint_integer(&param_server, exposure_name, 20, 32000, 10));

  // set parameter default values
  RCCHECK(rclc_parameter_set_int(&param_server, framerate_name, framerate));
  RCCHECK(rclc_parameter_set_int(&param_server, exposure_name, exposure));

  return true;
}

void destroy_entities()
{
  rmw_context_t * rmw_context = rcl_context_get_rmw_context(&support.context);
  (void) rmw_uros_set_context_entity_destroy_session_timeout(rmw_context, 0);
  rcl_publisher_fini(&imu_publisher, &node);
  rcl_publisher_fini(&baro_publisher, &node);
  rcl_subscription_fini(&subscriber, &node);
  rcl_timer_fini(&timer);
  rclc_parameter_server_fini(&param_server, &node);
  rclc_executor_fini(&executor);
  rcl_node_fini(&node);
  rclc_support_fini(&support);
}

```

hippo5329, 2024-11-10:
Try 

https://github.com/hippo5329/micro-ROS-demos-platformio/tree/rolling/rclc/parameter_server

jazzy esp32 seems work, pico crashed

```
ubuntu@alive-pug:~$ ros2 topic list
/micro_ros_pub
/parameter_events
/rosout
ubuntu@alive-pug:~$ ros2 param list
/micro_ros_node:
  publish_rate_ms
  publish_toogle
```

l4es, 2024-11-11:
Hi @hippo5329,

Thanks for your finding. Do you think that switching to use services seems to be a better solution by the way  ?

l4es, 2024-11-11:
Hi again @hippo5329,

I implemented and tested up to 2 services successfully together with 2 publishers, 1 subscription and 1 timer on the RP2040 Pico.

hippo5329, 2024-11-11:
I see RP2350/pico 2 is supported in platformio So micro-ROS port should work. We can try out. I just ordered an pico 2 to test.

https://github.com/maxgerhardt/platform-raspberrypi

l4es, 2024-11-23:
Hi @hippo5329,

I tried above _platformio.ini_ for the rp2350 but got the issue similar to this : https://github.com/micro-ROS/micro_ros_platformio/issues/153

hippo5329, 2024-11-28:
The latest earle update fixed this issue for pico2.

l4es, 2024-11-28:
Hi @hippo5329,

Thanks for the info. Did you mean the toolchain or the core? Currently I am still using the [ArduinoCore-Mbed](https://github.com/arduino/ArduinoCore-mbed) by default.

hippo5329, 2024-11-28:
I mean use the env I posted before,

```
[env:pico2]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = rpipico2
framework = arduino
board_build.core = earlephilhower
lib_ldf_mode = chain+
```

l4es, 2024-11-28:
Hi @hippo5329,

Thanks for the clarification. Using earlephilhower core would break the implementation of existing sensors libraries, but I will manage to update them.

closed 2024-11-28 by l4es

hippo5329, 2024-12-09:
I resolved this issue and created a pull-request.
https://github.com/earlephilhower/pico-quick-toolchain/pull/66

hippo5329, 2024-12-10:
```
# install toolchains
cd ~/micro_ros_arduino_examples_platformio/micro-ros_publisher
pio pkg install -e pico2
# patch time.h
cd ~
curl -O https://github.com/hippo5329/pico-quick-toolchain/raw/refs/heads/master/patches/lib-posix-timers.patch
cd ~/.platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include
patch -p4 < ~/lib-posix-timers.patch
# build and upload
cd ~/micro_ros_arduino_examples_platformio/micro-ros_publisher
pio run -e pico2 -t upload
```

hippo5329, 2024-12-11:
```
[env:pico2]
platform = https://github.com/hippo5329/platform-raspberrypi.git
board = rpipico2
board_build.core = earlephilhower
lib_ldf_mode = chain+
```

Only x86_64 host is supported at the moment.