micro-ROS/micro_ros_platformio #161: How to support RP2040 multicore ?

l4es, 2024-11-19:
Hi Forum,

We have the need of using the dual cores of the RP2040. PlatformIO documentation suggested to add board_build.core = earlephilhower into the platformio.ini but I faced with the following error : 

```
Compiling .pio/build/pico/lib0f7/micro_ros_platformio/platform_code/arduino/clock_gettime.cpp.o
src/main.cpp:9:10: fatal error: Arduino.h: No such file or directory

*****************************************************************
* Looking for Arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Arduino.h"
* Web  > https://registry.platformio.org/search?q=header:Arduino.h
*
*****************************************************************

    9 | #include <Arduino.h>
      |          ^~~~~~~~~~~
compilation terminated.
Compiling .pio/build/pico/lib0f7/micro_ros_platformio/platform_code/arduino/serial/micro_ros_transport.cpp.o
.pio/libdeps/pico/micro_ros_platformio/platform_code/arduino/clock_gettime.cpp:1:10: fatal error: Arduino.h: No such file or directory

*****************************************************************
* Looking for Arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Arduino.h"
* Web  > https://registry.platformio.org/search?q=header:Arduino.h
*
*****************************************************************

    1 | #include <Arduino.h>
      |          ^~~~~~~~~~~
compilation terminated.
Compiling .pio/build/pico/libe26/Wire/Wire.cpp.o
.pio/libdeps/pico/micro_ros_platformio/platform_code/arduino/serial/micro_ros_transport.cpp:1:10: fatal error: Arduino.h: No such file or directory

*****************************************************************
* Looking for Arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Arduino.h"
* Web  > https://registry.platformio.org/search?q=header:Arduino.h
*
*****************************************************************

    1 | #include <Arduino.h>
      |          ^~~~~~~~~~~
compilation terminated.
In file included from /root/.platformio/packages/framework-arduino-mbed/libraries/Wire/Wire.cpp:23:
/root/.platformio/packages/framework-arduino-mbed/libraries/Wire/Wire.h:21:10: fatal error: Arduino.h: No such file or directory

*****************************************************************
* Looking for Arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Arduino.h"
* Web  > https://registry.platformio.org/search?q=header:Arduino.h
*
*****************************************************************

   21 | #include "Arduino.h"
      |          ^~~~~~~~~~~
compilation terminated.
*** [.pio/build/pico/src/main.cpp.o] Error 1
*** [.pio/build/pico/lib0f7/micro_ros_platformio/platform_code/arduino/clock_gettime.cpp.o] Error 1
*** [.pio/build/pico/lib0f7/micro_ros_platformio/platform_code/arduino/serial/micro_ros_transport.cpp.o] Error 1
*** [.pio/build/pico/libe26/Wire/Wire.cpp.o] Error 1
====================================================================================== [FAILED] Took 321.49 seconds ======================================================================================
```

Without adding above line into the platformio.ini, the code was compiled without any problem but it got use only the first core of RP2040.

Thanks in advance and best regards,
Khang

hippo5329, 2024-11-19:
The latest updates of earle toolchain caused the issue. You may revert to earlier toolchain.

```
[env:pico_el]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git#196d31bbafaf60b84751b1a415d8dca2365debdf
board = rpipico
lib_ldf_mode = chain+
```

l4es, 2024-11-20:
Hi @hippo5329,

Thanks for your advice. Just make sure that I understand well, here below my current platformio.ini : 

```
[env]
framework = arduino
lib_deps = 
    https://github.com/micro-ROS/micro_ros_platformio
    Wire
    SPI
    robtillaart/MS5611@^0.4.0
    finani/ICM42688@^1.1.0
    paulstoffregen/Time@^1.6.1

[env:pico]
platform = raspberrypi
board = pico
upload_protocol = picotool

; Flash Size: 2MB (no filesystem)
board_upload.maximum_size = 2097152

; CPU Speed: 133MHz
board_build.f_cpu =  133000000L

; Disable Debug and C++ Exceptions ; Optimize for size (Small OS)
build_unflags = -DDEBUG -fexceptions
build_flags = -DNDEBUG -fno-exceptions -Os -ffunction-sections -fdata-sections
```
In order to add  board_build.core = earlephilhower  and compile without error, I should modify my platformio.ini as below  : 

```
diff --git a/workspace/platformio.ini b/workspace/platformio.ini
index 2f65923..9537275 100644
--- a/workspace/platformio.ini
+++ b/workspace/platformio.ini
@@ -18,11 +18,10 @@ lib_deps =
     finani/ICM42688@^1.1.0
     paulstoffregen/Time@^1.6.1
 
-board_microros_user_meta = rtr_inertial_microros.meta
-
-[env:pico]
-platform = raspberrypi
-board = pico
+[env:pico_el]
+platform = https://github.com/maxgerhardt/platform-raspberrypi.git#196d31bbafaf60b84751b1a415d8dca2365debdf
+board = rpipico
+lib_ldf_mode = chain+
 upload_protocol = picotool
 
 ; Flash Size: 2MB (no filesystem)
@@ -30,6 +29,7 @@ board_upload.maximum_size = 2097152
 
 ; CPU Speed: 133MHz
 board_build.f_cpu =  133000000L
+board_build.core = earlephilhower
 
 ; Disable Debug and C++ Exceptions ; Optimize for size (Small OS)
 build_unflags = -DDEBUG -fexceptions
```

and would be able to use the dual cores of the RP2040, is that correct ?

Thanks and best regards,
Khang

hippo5329, 2024-11-20:
"board_build.core = earlephilhower" should be added for more boards support.