micro-ROS/micro_ros_platformio #163: Trouble Connecting to Ubuntu via Teensy 4.1 Native Ethernet

PrestonBodak, 2024-12-07:
Hello all,

I'm new to Teensy and ROS2 development, my setup is as follows:

- Hardware: Teensy 4.1 w/ Native Ethernet
- Software: Ubuntu 22.04.5 LTS, PlatformIO (VSCode)
- Version: Humble

I am currently trying to connect to ROS2 on an Ubuntu environment via the Teensy 4.1 Ethernet kit. So far, I have been able to use the standard [micro-ros_publisher example](https://github.com/micro-ROS/micro_ros_platformio/blob/main/examples/micro-ros_publisher/src/Main.cpp) to connect and spin the starter node etc. However, upon switching to the native ethernet transport, I can't seem to make the connection.

My current Teensy code can be found here (a modification of the [micro_ros-arduino ethernet publisher](https://github.com/micro-ROS/micro_ros_arduino/blob/humble/examples/micro-ros_publisher_ethernet/micro-ros_publisher_ethernet.ino)): https://pastebin.com/FBjWCKnz
My current platform.ini setup is as follows:

```; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
board_microros_distro = humble
board_microros_transport = native_ethernet
lib_deps = 
	ssilverman/QNEthernet@^0.29.1
	http://github.com/micro-ROS/micro_ros_platformio
	;mirs240x/micro_ros_arduino@^2.0.7-humble
```

#### Steps to reproduce the issue
First, I launch the micro-ros agent in the Ubuntu environment via docker with:
```
sudo docker run -it --rm -v /dev:/dev -v /dev/shm:/dev/shm --privileged --net=host microros/micro-ros-agent:$ROS_DISTRO udp4 --port 2525 -v6
```

I then upload my code to the Teensy board, where the code appears to enter the error loop on line 73 at
```
RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));
```

At this point, the agent has produced the following results before stopping:
[image]

#### Expected behavior

Upon starting the Teensy, I anticipate seeing an agent log with a datawriter being created and an increment between each recv_message as seen here when using the serial example:

[image]
[image]

Using rqt with this example allows me to see the micro_ros_platformio node, indicating that the data makes it completely through the middleware.

I am hoping that the "session re-established" logs indicate that I may just be missing something simple. Any help would be greatly appreciated, and I am happy to provide more information!

hippo5329, 2024-12-08:
I would not suggest micro-ros agent docker in this case. You should build the agent.

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki#build-micro-ros-agent

hippo5329, 2024-12-08:
You should specify network setting in

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/blob/main/platformio/platformio.ini#L11

```
build_flags =
    '-D LOCAL_MAC={ 0xAA, 0xBB, 0xCC, 0xEE, 0xDD, 0xFF }'
    '-D LOCAL_IP={ 192, 168, 1, 101 }'
    '-D AGENT_IP={ 192, 168, 1, 100 }'
    '-D AGENT_PORT=8888'
```

The IP and port must match the agent.

hippo5329, 2024-12-08:
Your publisher code is wrong. You may follow the wiki for basic publisher and re-connection examples. Ethernet transport and wifi transport are very similar. Both are udp4. The agent sides are the same.

hippo5329, 2024-12-08:
pio run -e teensy41_eth -t upload

PrestonBodak, 2024-12-08:
I appreciate your help,

When attempting to build the agent, I receive the following errors:

[image]

This appeared to indicate that I did not have FastDDS properly installed, so I [built it from source via colcon (3.1.3)](https://fast-dds.docs.eprosima.com/en/latest/installation/sources/sources_linux.html). Following this, I verified that the file does exist at /opt/ros/humble/include/fastrtps/fastdds/rtps/common/Guid.h, yet the error persists when attempting to build.

With regards to my platform.ini, I have added [FreeRTOS](https://github.com/juliandesvignes/FreeRTOS-Teensy4) to my lib_deps as well as:
```
lib_ldf_mode = chain+
build_flags = 
    '-D LOCAL_IP={ (IP omitted) }'
    '-D AGENT_IP={ (IP omitted) }'
    '-D AGENT_PORT=2525'
```

My setup code was updated in accordance with the network setting change (with the mac still being read dynamically):
```
IPAddress arduino_ip(LOCAL_IP);
IPAddress agent_ip(AGENT_IP);
set_microros_native_ethernet_transports(arduino_mac, arduino_ip, agent_ip, AGENT_PORT);
```

Apart from this, [my updated publisher code](https://pastebin.com/CNwXarus) (now following the timer/executor format) appears quite similar to [this example](https://github.com/hippo5329/micro_ros_arduino_examples_platformio/blob/main/micro-ros_publisher/micro-ros_publisher.ino); is there somewhere else on the wiki I should be referencing? Following these changes, the serial example still uploads and runs successfully with the docker agent, but my ethernet code exhibits the same behavior as before.

hippo5329, 2024-12-09:
I launched an LXD container and built the agent without error. You may try with an container or VM. You can try the basic publisher and re-connection examples in my repo. There is no need to modify the codes. I ordered a teensy41 with ethernet add-on to test. I will share with you later.

hippo5329, 2024-12-09:
The delay(100) in the loop() should be removed.

hippo5329, 2024-12-13:
I tested the publisher and re-connection examples on a teensy41 ethernet. Both work.

    pio run -e teensy41_eth -t upload

```
ubuntu@x870rs:~$ ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888
[1734069663.574744] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1734069663.574857] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1734069663.672834] info     | Root.cpp           | create_client            | create                 | client_key: 0x269CE89B, session_id: 0x81
[1734069663.672888] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x269CE89B, address: 10.2.0.91:47138
[1734069663.677972] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x269CE89B, participant_id: 0x000(1)
[1734069663.678833] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x269CE89B, topic_id: 0x000(2), participant_id: 0x000(1)
[1734069663.679861] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x269CE89B, publisher_id: 0x000(3), participant_id: 0x000(1)
[1734069663.680961] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x269CE89B, datawriter_id: 0x000(5), publisher_id: 0x000(3)
```

You should run example - native ethernet- web server to verify your hardware works.

PrestonBodak, 2025-01-08:
After some time away from the project, I have returned and checked my network settings. It turns out, my agent IP address was not properly configured. The code works after fixing the IP address!

closed 2025-01-08 by PrestonBodak