micro-ROS/micro_ros_platformio #164: MicroRos message type

mehtivakili, 2024-12-27:
I can create my custom message type on micro-ros platformIO . Is that possible or compatible with the platformIO version of that?
I appreciate any help..

hippo5329, 2024-12-27:
Please describe your problem in more details.

mehtivakili, 2024-12-30:
@hippo5329 
I want to create my custom messages on microros using platformIO and there is no straight introduction on the documentation nor the code how can I use mircro-ros and ploatformIO implementation of that to do this? I think I should recompile that or something like that? but what should I recompile or build? the downloaded package on the platformIO library"microros-platformIO" or the "ros2" on my own system?
Thanks for quick respond!

hippo5329, 2024-12-30:
Please check the wiki,

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki#custom-message-type

mehtivakili, 2025-01-04:
@hippo5329 
thanks man!
But I unfortunately found that part previously and it didn't help me find out how can I compile or build that message 
after creating that message type :(

hippo5329, 2025-01-06:
Did you follow the tutorial to create your custom message?

https://docs.ros.org/en/jazzy/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html

hippo5329, 2025-01-06:
Custom message examples,

https://github.com/micro-ROS/micro_ros_platformio/blob/main/ci/src/main.cpp

https://github.com/hippo5329/micro-ROS-demos-platformio/blob/rolling/rclc/complex_msg_publisher/main.c

wiki
https://github.com/hippo5329/micro-ROS-demos-platformio/wiki

mehtivakili, 2025-01-15:
@hippo5329 thanks for your respond, I still couldn't check out the repos you've provided but I'll look forward to use them as soon as possible bro

hippo5329, 2025-01-15:
The links to examples code are files. You will need to checkout the repo instead. Follow the wiki,

```
git clone https://github.com/hippo5329/micro-ROS-demos-platformio.git
cd micro-ROS-demos-platformio/rclc/complex_msg_publisher
pio run -e esp32 -t upload
```