micro-ROS/micro_ros_platformio #166: Subscribing to custom message with an array of int's

GerardHarkema, 2025-01-13:
Hello,

 I created a custom message with an array of uint32. 

Message definition:
_uint32[] dummy_

In my microros application(on a ESP32) i subscribe to the topic created from this message.
When publishing on this topic by command line:

_ros2 topic pub -r 1 /my_topic "{dummy: [1, 2]}"_

the function _rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100))_ returns 1 and the callback is not executed.

But when i publish an empty array the callback is executed.
_ros2 topic pub -r 1 /my_topic "{dummy: []}"_

Does anybody know if array's on subscribers are tested? I'am also using an array as publisher in my project, this works fine.

Does anybody know an solution, thanks in advance.

Kind regards!

hippo5329, 2025-01-13:
You will need to allocate variable size message.

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki#string-array-multiarray

hippo5329, 2025-01-13:
For example,

https://github.com/hippo5329/linorobot2_hardware/blob/master/firmware/src/firmware.cpp#L563

GerardHarkema, 2025-01-14:
Hi Hippo,

Thanks, this helped a lot.

Kind regards
Gerard

closed 2025-01-14 by GerardHarkema