micro-ROS/micro_ros_platformio #168: Missing tf2_msgs/msg/* topic headers

xopxe, 2025-01-23:
I'm using an ESP32-PICO-Kit board, with the following description in the platformio.ini:

```ini
[env:pico32]
platform = espressif32
board = pico32
framework = arduino
board_microros_distro = jazzy  
board_microros_transport = serial
```

I'm compiling using VSCode. I've successfully published topics of various types, such as temperature, imu, odometry, etc. For some reason, the `tf2_msgs` are not in place. As an example, here is my `odometry.h` header (after compiling the project):
```sh
my_project_folder$ find . -name odometry.h
./.pio/libdeps/pico32/micro_ros_platformio/libmicroros/include/nav_msgs/msg/odometry.h
./.pio/libdeps/pico32/micro_ros_platformio/build/mcu/build/nav_msgs/rosidl_generator_c/nav_msgs/msg/odometry.h
./.pio/libdeps/pico32/micro_ros_platformio/build/mcu/install/include/nav_msgs/nav_msgs/msg/odometry.h
```
But there's no `.pio/libdeps/pico32/micro_ros_platformio/libmicroros/include/tf2_msgs` folder. This folder with its content does exist in the "pure Arduino" version of micro-ros ([here](https://github.com/micro-ROS/micro_ros_arduino/tree/jazzy/src/tf2_msgs)).

Is it supposed to exist for micro_ros_platformio?

pablogs9, 2025-01-23:
If I'm not wrong, it is not included by default. But you can include packages: https://github.com/micro-ROS/micro_ros_platformio?tab=readme-ov-file#extra-packages

xopxe, 2025-01-23:
Thanks, it was so.

I cloned [this](https://github.com/ros2/geometry2.git)  repository and checked out the branch of interest (jazzy in my case). Then, I manually copied the `tf2_msg` folder (which contains a package.xml) into an `extra_packages` in my project. 
Finally, deleted the .pio folder and rebuilt everything. 

I suppose this can be automated somehow.

closed 2025-01-23 by xopxe