micro-ROS/micro_ros_platformio #174: Adding library to PlatformIO on Mac fails

Maximilian-Miller, 2025-04-03:
## Adding microROS library to PlatformIO fails during build process

Hi, I'm trying to add the microROS library to a blank platformIO project, but the build of the library fails. I added all necessary software modules (git, cmake....) and installed the binutils as described in the requirements.

- Hardware description: Host: Apple MacBook Pro M4 Pro, MacOs 15.3 Target: ESP32-S3 DevKitM-1
- Installation type: microROS
- Version or commit hash: jazzy

#### Steps to reproduce the issue
- create new platformIO project
- add lib to platform.ini 
`lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio`
- wait for build to fail
- the same errors occur if the build process of the project is started 

#### Expected behavior
Library builded and added to project

#### Actual behavior
Build fails

#### Additional information
Console output:
```bash
Resolving esp32-s3-devkitm-1 dependencies...
Library Manager: Installing git+https://github.com/micro-ROS/micro_ros_platformio
git version 2.39.5 (Apple Git-154)
Cloning into '/Users/maximilian/.platformio/.cache/tmp/pkg-installing-xjp47jz0'...

Library Manager: micro_ros_platformio@0.0.1+sha.f474915 has been installed!
Updating metadata for the vscode IDE...
UserSideException: Processing esp32-s3-devkitm-1 (platform: espressif32; board: esp32-s3-devkitm-1; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitm-1.html
PLATFORM: Espressif 32 (54.3.20+develop.sha.fafc700) > Espressif ESP32-S3-DevKitM-1
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.2.0 
 - framework-arduinoespressif32-libs @ 5.4.0+sha.2f7dcd862a 
 - tool-esptoolpy @ 4.8.9 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-riscv32-esp @ 14.2.0+20241119 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing importlib-resources with pip at PlatformIO environment
/Users/maximilian/.platformio/penv/bin/python -m pip install importlib-resources
Requirement already satisfied: importlib-resources in /Users/maximilian/.platformio/penv/lib/python3.11/site-packages (6.5.2)
Installing pyyaml with pip at PlatformIO environment
/Users/maximilian/.platformio/penv/bin/python -m pip install pyyaml
Requirement already satisfied: pyyaml in /Users/maximilian/.platformio/penv/lib/python3.11/site-packages (6.0.2)
Installing markupsafe==2.0.1 with pip at PlatformIO environment
/Users/maximilian/.platformio/penv/bin/python -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in /Users/maximilian/.platformio/penv/lib/python3.11/site-packages (2.0.1)
Installing empy==3.3.4 with pip at PlatformIO environment
/Users/maximilian/.platformio/penv/bin/python -m pip install empy==3.3.4
Requirement already satisfied: empy==3.3.4 in /Users/maximilian/.platformio/penv/lib/python3.11/site-packages (3.3.4)
Configuring esp32-s3-devkitm-1 with transport serial
Downloading micro-ROS dev dependencies
	 - Downloaded ament_cmake
	 - Downloaded ament_lint
	 - Downloaded ament_package
	 - Downloaded googletest
	 - Downloaded ament_cmake_ros
	 - Downloaded ament_index
[... 378 lines cut ...]
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `/Users/maximilian/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc  -I/Users/maximilian/Desktop/ESP32-MicroROS-PIO-Test/.pio/libdeps/esp32-s3-devkitm-1/micro_ros_platformio/build/mcu/src/micro-CDR/include -I/Users/maximilian/Desktop/ESP32-MicroROS-PIO-Test/.pio/libdeps/esp32-s3-devkitm-1/micro_ros_platformio/build/mcu/build/microcdr/include -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -std=gnu17 -Wno-old-style-declaration -Wno-strict-prototypes -Werror=return-type -Os -mlongcalls -ffunction-sections -fdata-sections -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wno-error=extra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -mdisable-hardware-atomics -freorder-blocks -Wwrite-strings -fstack-protector -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion ;-DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=<lib/address_resolve/AddressResolve_DefaultImpl.h>; -MMD -DCLOCK_MONOTONIC=0 -D'__attribute__(x)=' -O3 -DNDEBUG -std=gnu99 -fPIC -Wall -Wextra -Wshadow -pedantic -Wcast-align -Wunused -Wconversion -Wnull-dereference -Wduplicated-branches -Wduplicated-cond -Wrestrict -Wdouble-promotion -Wsign-conversion -fdata-sections -ffunction-sections -MD -MT CMakeFiles/microcdr.dir/src/c/types/string.c.obj -MF CMakeFiles/microcdr.dir/src/c/types/string.c.obj.d -o CMakeFiles/microcdr.dir/src/c/types/string.c.obj -c /Users/maximilian/Desktop/ESP32-MicroROS-PIO-Test/.pio/libdeps/esp32-s3-devkitm-1/micro_ros_platformio/build/mcu/src/micro-CDR/src/c/types/string.c'
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `/Users/maximilian/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc  -I/Users/maximilian/Desktop/ESP32-MicroROS-PIO-Test/.pio/libdeps/esp32-s3-devkitm-1/micro_ros_platformio/build/mcu/src/micro-CDR/include -I/Users/maximilian/Desktop/ESP32-MicroROS-PIO-Test/.pio/libdeps/esp32-s3-devkitm-1/micro_ros_platformio/build/mcu/build/microcdr/include -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -std=gnu17 -Wno-old-style-declaration -Wno-strict-prototypes -Werror=return-type -Os -mlongcalls -ffunction-sections -fdata-sections -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wno-error=extra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -mdisable-hardware-atomics -freorder-blocks -Wwrite-strings -fstack-protector -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion ;-DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=<lib/address_resolve/AddressResolve_DefaultImpl.h>; -MMD -DCLOCK_MONOTONIC=0 -D'__attribute__(x)=' -O3 -DNDEBUG -std=gnu99 -fPIC -Wall -Wextra -Wshadow -pedantic -Wcast-align -Wunused -Wconversion -Wnull-dereference -Wduplicated-branches -Wduplicated-cond -Wrestrict -Wdouble-promotion -Wsign-conversion -fdata-sections -ffunction-sections -MD -MT CMakeFiles/microcdr.dir/src/c/types/basic.c.obj -MF CMakeFiles/microcdr.dir/src/c/types/basic.c.obj.d -o CMakeFiles/microcdr.dir/src/c/types/basic.c.obj -c /Users/maximilian/Desktop/ESP32-MicroROS-PIO-Test/.pio/libdeps/esp32-s3-devkitm-1/micro_ros_platformio/build/mcu/src/micro-CDR/src/c/types/basic.c'
make[5]: *** [CMakeFiles/microcdr.dir/src/c/common.c.obj] Error 2
make[5]: *** Waiting for unfinished jobs....
make[5]: *** [CMakeFiles/microcdr.dir/src/c/types/string.c.obj] Error 2
make[5]: *** [CMakeFiles/microcdr.dir/src/c/types/basic.c.obj] Error 2
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `/Users/maximilian/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc  -I/Users/maximilian/Desktop/ESP32-MicroROS-PIO-Test/.pio/libdeps/esp32-s3-devkitm-1/micro_ros_platformio/build/mcu/src/micro-CDR/include -I/Users/maximilian/Desktop/ESP32-MicroROS-PIO-Test/.pio/libdeps/esp32-s3-devkitm-1/micro_ros_platformio/build/mcu/build/microcdr/include -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -std=gnu17 -Wno-old-style-declaration -Wno-strict-prototypes -Werror=return-type -Os -mlongcalls -ffunction-sections -fdata-sections -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wno-error=extra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -mdisable-hardware-atomics -freorder-blocks -Wwrite-strings -fstack-protector -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion ;-DCHIP_A [... cut ...] evkitm-1/micro_ros_platformio/build/mcu/src/micro-CDR/src/c/types/sequence.c'
make[5]: *** [CMakeFiles/microcdr.dir/src/c/types/sequence.c.obj] Error 2
make[4]: *** [CMakeFiles/microcdr.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [ucdr-prefix/src/ucdr-stamp/ucdr-build] Error 2
make[1]: *** [CMakeFiles/ucdr.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< microcdr [1.22s, exited with code 2]

========================= [FAILED] Took 31.80 seconds =========================
```

soylentOrange, 2025-05-21:
Same for me! 
@Maximilian-Miller did you find a solution?

Maximilian-Miller, 2025-05-22:
I got it working and can now compile the micro-ros library on my Mac. The issue was that I had installed the platform-espressif32 from pioarduino. As soon as that platform is installed, the CMake error appears. Using the original Espressif platform resolves the issue.

soylentOrange, 2025-05-24:
@Maximilian-Miller Thanks for the feedback! After setting `platform = espressif32@6.10.0` in my platformio.ini it worked for me as well. A pity that I'm limited to using arduino 2 now...