micro-ROS/micro_ros_platformio #177: Errors building for rp2040/2350: `clock_gettime` / `__atomic_test_and_set`

snaens, 2025-05-22:
## Issue template

- Hardware description: pi pico * / rp2040 / rp2350
- Installation type: platformio CLI
- Version or commit hash: microros jazzy

#### Steps to reproduce the issue
Try to build a simple script for rp2040 or rp2350, using…

1. https://github.com/maxgerhardt/platform-raspberrypi.git#1ab176cd4d4edd38a156e0c25e12e86ad96b21ef
2. https://github.com/maxgerhardt/platform-raspberrypi.git
3. https://github.com/hippo5329/platform-raspberrypi.git

**Platformio.ini:**
```ini
[env:pico]
platform = <platform link>
board = rpipico <or rpipico2>
framework = arduino
board_build.core = earlephilhower

board_microros_distro = jazzy
board_microros_transport = serial
lib_ldf_mode = chain+

lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio
```

**Code:**
https://github.com/micro-ROS/micro_ros_platformio/blob/main/examples/micro-ros_publisher/src/Main.cpp

**Procedure:**
3 Docker debian containers, each with a fresh platformio install.
<details><summary>Dockerfile</summary>
 

```dockerfile
FROM debian:bookworm

RUN apt update && apt install \
      git python3-venv wget nano gcc-arm-none-eabi cmake g++ gcc -y \
      && rm -rf /var/cache/apt/

RUN wget -O get-platformio.py https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py &&\
    python3 get-platformio.py

RUN mkdir -p /usr/local/bin &&\
    ln -s ~/.platformio/penv/bin/platformio /usr/local/bin/platformio &&\
    ln -s ~/.platformio/penv/bin/pio /usr/local/bin/pio &&\
    ln -s ~/.platformio/penv/bin/piodebuggdb /usr/local/bin/piodebuggdb &&\
    mkdir -p /etc/udev/rules.d/ &&\
    wget -O /etc/udev/rules.d/99-platformio-udev.rules https://raw.githubusercontent.com/platformio/platformio-core/develop/platformio/assets/system/99-platformio-udev.rules

ENTRYPOINT ["/bin/bash"]
```

</details>

**run:**
```bash
git clone https://github.com/micro-ROS/micro_ros_platformio/
cd micro_ros_platformio/examples/micro-ros_publisher/ #use aforementioned example 
nano -n platformio.ini #replace with my `platformio.ini`
# each container uses one of the 3 platform links
pio run
```

#### Expected behavior
All 3 platforms should build fine.
[... 224 lines cut ...]
/root/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /root/.platformio/packages/framework-arduinopico/lib/rp2040/libpico.a(atomic.c.o): in function `__atomic_store_8':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/src/rp2_common/pico_atomic/atomic.c:132: multiple definition of `__atomic_store_8'; .pio/libdeps/pico/micro_ros_platformio/libmicroros/libmicroros.a(librcutils__atomic_64bits.c.obj):atomic_64bits.c:(.text.__atomic_store_8+0x0): first defined here
/root/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /root/.platformio/packages/framework-arduinopico/lib/rp2040/libpico.a(atomic.c.o): in function `__atomic_exchange_8':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/src/rp2_common/pico_atomic/atomic.c:163: multiple definition of `__atomic_exchange_8'; .pio/libdeps/pico/micro_ros_platformio/libmicroros/libmicroros.a(librcutils__atomic_64bits.c.obj):atomic_64bits.c:(.text.__atomic_exchange_8+0x0): first defined here
/root/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /root/.platformio/packages/framework-arduinopico/lib/rp2040/libpico.a(atomic.c.o): in function `__atomic_fetch_add_8':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/src/rp2_common/pico_atomic/atomic.c:189: multiple definition of `__atomic_fetch_add_8'; .pio/libdeps/pico/micro_ros_platformio/libmicroros/libmicroros.a(librcutils__atomic_64bits.c.obj):atomic_64bits.c:(.text.__atomic_fetch_add_8+0x0): first defined here
/root/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: .pio/libdeps/pico/micro_ros_platformio/libmicroros/libmicroros.a(librcutils__atomic_64bits.c.obj): in function `__atomic_load_8':
atomic_64bits.c:(.text.__atomic_load_8+0x2a): undefined reference to `__atomic_test_and_set'
/root/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: .pio/libdeps/pico/micro_ros_platformio/libmicroros/libmicroros.a(librcutils__atomic_64bits.c.obj): in function `__atomic_store_8':
atomic_64bits.c:(.text.__atomic_store_8+0x2e): undefined reference to `__atomic_test_and_set'
/root/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: .pio/libdeps/pico/micro_ros_platformio/libmicroros/libmicroros.a(librcutils__atomic_64bits.c.obj): in function `__atomic_exchange_8':
atomic_64bits.c:(.text.__atomic_exchange_8+0x2e): undefined reference to `__atomic_test_and_set'
collect2: error: ld returned 1 exit status
*** [.pio/build/pico/firmware.elf] Error 1
```

</details>

[log_3.txt](https://github.com/user-attachments/files/20398274/log_3.txt)

rp2040 and rp2350 fail in the exact same fashion.

#### Additional information

Unfortunately I cannot use the older commit version, as it is lacking features I need :(

snaens, 2025-05-22:
After writing this down I just thought of a workaround for case 3: `https://github.com/hippo5329/platform-raspberrypi.git`

Once compiled for the first time, run

```
arm-none-eabi-ar dv .pio/libdeps/pico/micro_ros_platformio/libmicroros/libmicroros.a librcutils__atomic_64bits.c.obj
```

<details>
<summary>Full log</summary>

```console
root@container:~/micro_ros_platformio/examples/micro-ros_publisher# arm-none-eabi-ar dv .pio/libdeps/pico/micro_ros_platformio/libmicroros/libmicroros.a librcutils__atomic_64bits.c.obj
d - librcutils__atomic_64bits.c.obj

root@container:~/micro_ros_platformio/examples/micro-ros_publisher# pio run
Processing pico (platform: https://github.com/hippo5329/platform-raspberrypi.git; board: rpipico2; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/rpipico2.html
PLATFORM: Raspberry Pi RP2040 (1.15.0+sha.e95c19c) > Pico 2
HARDWARE: RP2350 150MHz, 512KB RAM, 4MB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.40504.0+sha.7dcfe4d 
 - tool-picotool-rp2040-earlephilhower @ 5.140200.240929 (14.2.0) 
 - toolchain-rp2040-earlephilhower @ 5.140200.241213 (14.2.0)
Flash size: 4.00MB
Sketch size: 4.00MB
Filesystem size: 0.00MB
PSRAM size: 0.00MB
Maximium Sketch size: 4190208 EEPROM start: 0x103ff000 Filesystem start: 0x103ff000 Filesystem end: 0x103ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ soft
Installing importlib-resources with pip at PlatformIO environment
/root/.platformio/penv/bin/python -m pip install importlib-resources
Requirement already satisfied: importlib-resources in /root/.platformio/penv/lib/python3.11/site-packages (6.5.2)
Installing pyyaml with pip at PlatformIO environment
/root/.platformio/penv/bin/python -m pip install pyyaml
Requirement already satisfied: pyyaml in /root/.platformio/penv/lib/python3.11/site-packages (6.0.2)
Installing markupsafe==2.0.1 with pip at PlatformIO environment
/root/.platformio/penv/bin/python -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in /root/.platformio/penv/lib/python3.11/site-packages (2.0.1)
Installing empy==3.3.4 with pip at PlatformIO environment
/root/.platformio/penv/bin/python -m pip install empy==3.3.4
Requirement already satisfied: empy==3.3.4 in /root/.platformio/penv/lib/python3.11/site-packages (3.3.4)
Configuring rpipico2 with transport serial
micro-ROS already built
Found 69 compatible libraries
Scanning dependencies...
Dependency Graph
|-- micro_ros_platformio @ 0.0.1+sha.f474915
Building in release mode
Linking .pio/build/pico/firmware.elf
Generating UF2 image
picotool uf2 convert -t elf ".pio/build/pico/firmware.elf" ".pio/build/pico/firmware.uf2"
Retrieving maximum program size .pio/build/pico/firmware.elf
Flash size: 4.00MB
Sketch size: 4.00MB
Filesystem size: 0.00MB
PSRAM size: 0.00MB
Maximium Sketch size: 4190208 EEPROM start: 0x103ff000 Filesystem start: 0x103ff000 Filesystem end: 0x103ff000
Checking size .pio/build/pico/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   6.1% (used 31904 bytes from 524288 bytes)
Flash: [          ]   3.1% (used 128392 bytes from 4190208 bytes)
Building .pio/build/pico/firmware.bin
Building .pio/build/pico/firmware.bin.signed
============================================================ [SUCCESS] Took 6.55 seconds ============================================================
```

</details>

This removes the clashing `librcutils__atomic_64bits.c.obj`.
No Idea if this is a good idea, but hey, the example topic works.

It results in a successful build.

snaens, 2025-05-22:
Oh, and this is relevant:
https://github.com/earlephilhower/arduino-pico/discussions/2497

hippo5329, 2025-05-23:
Yes, there are two known issues, the atomic ops and clock_gettime().

You may follow the wiki. It fixed both pico and pico2.

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

snaens, 2025-05-23:
The wiki includes `platformio.ini` content for stock pico, pico with Earle's core and pico2 with Earle's core:

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/blob/7fc84e616d0a8e3fbadbacfd2db285e09824e996/platformio/platformio.ini#L77-L92

Following the instructions in fresh containers (literally copying them):
- Stock builds fine
- pico_el fails and can be resolved with my workaround
  (installing `gcc-arm-none-eabi` first, and change the path to `pico_el`) 
- pico2 fails because the [hippo5329/pico-quick-toolchain](https://github.com/hippo5329/pico-quick-toolchain) repo has disappeared. (did you delete it yesterday?)

hippo5329, 2025-05-23:
Oh.. Sorry, I mistakenly removed the pico_quick_toolchain yesterday during my Spring Cleanup. I will rebuild the toolchain later.

snaens, 2025-05-23:
I reran the test for pico 1 and 2, 
using `platform = https://github.com/hippo5329/platform-raspberrypi.git` from a cached copy I had.

You assumed correctly; pico2 builds fine, only pico1 fails (how did I miss this).
So the issue is with pico-sdk internally, correct?
fantastic 😒

hippo5329, 2025-05-26:
For pico2, the toolchains in my repo are updated to gcc 14.3. Please try out.

I did not test pico. I think the atomic ops issue persist.

hippo5329, 2025-06-02:
@snaens Atomic ops issues of pico resolved with 

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/commit/f7a15e28fbe2de26b6932814b259b229d011df5b

maxgerhardt/platform-raspberrypi will be used when it is updated to earlephilhower/pico-quick-toolchain 4.1.0 .

closed 2026-11-07 by snaens